roboticslab-uc3m/teo-openrave-models #3: Think about openrave-database-generators

jgvictores, 2017-03-02:
Think about [http://openrave.org/docs/latest_stable/databases/#openrave-database-generators](openrave-database-generators)

jgvictores, 2017-05-16:
This issue is becoming increasingly necessary (e.g. https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/99).

jgvictores, 2017-05-16:
More specifically: http://openrave.org/docs/latest_stable/openravepy/databases.kinematicreachability/

rsantos88, 2017-05-17:
This is the reachability of Teo's right arm with the original model:
[image]
To know if it's considering the joint limits, I've tried to remove all the`<limitsdeg>` tags along the `teo_rightArm.kinbody.xml` file, but when I generate the reachability again, it shows the same result...
[image]

jgvictores, 2017-05-17:
@rsantos88 Looks great, thanks!

1. Could you specify the command line you used to generate this output?
1. Is there any output file generated? Does it seem useful or somehow human readable? Maybe you could upload it somewhere or copy+paste a fragment?

rsantos88, 2017-05-17:
Yes, the command line used is:

`openrave.py --database kinematicreachability --robot=/home/teo/repos/teo-openrave-models/openrave/teo/teo.robot.xml --numthreads=4 --manipname=rightArm`

This will take some time to generate the reachability... you've to wait...
Then, it will create some files in `/home/user/.openrave/`. When the process finishes, you can see the result writing the same line adding `show` parameter:

`openrave.py --database kinematicreachability --robot=/home/teo/repos/teo-openrave-models/openrave/teo/teo.robot.xml --numthreads=4 --manipname=rightArm --show`

Remember that you need to install *mayavi2* if you don't have installed it.

rsantos88, 2017-05-17:
Ok, possibly this is the solution to the problem ([see](http://openrave-users-list.185357.n3.nabble.com/Problem-using-Joint-Limits-td4026965.html))

rsantos88, 2017-05-17:
Changed the "circular" joints to false, the reachability result is...
[image]

jgvictores, 2017-05-18:
Great job! Could you commit the updated model?

It seems like an improvement. Would this model work with `teoSim` etc?

rsantos88, 2017-05-19:
The limits works well apparently. However, when you exceed the limits in teoSim, openrave warns to you with a message like this: `[kinbody.cpp:1515 SetDOFValues] dof 4 value is not in limits 1.566438e+00<1.221730e+00` and the model stops respecting the limit configured in the xml, but if you want to know the actual position, it returns the extraimited position value. 
This problem is referenced at [ [YarpOpenraveControlboard] could get limits from openrave](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/12)] issue. 
Even so, @jgvictores do you want to update the model with this changes?
The changes are done in (b1db54994f409bb04bdb5cd845c8d370cc501add)

jgvictores, 2017-05-25:
I'm not a big fan of duplicate files, but stranded branches and open issues do not help either. I'm strongly considering having both versions with different names on `develop`.

jgvictores, 2017-06-26:
@rsantos88 Do you think you could create 2 more manipulators? Something like `fullRightArm` and `fullLeftArm` that include the trunk (initially, including all joint limits).

rsantos88, 2017-07-03:
Ok @jgvictores , this is the visual result after calculate the reachability with the corrections of (#4) issue:
[image]
Here you have other nice top view:
[image]
I've added these two other manipulators in a different [branch](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/reachabilityWithTrunk). If you want, we can merge it with `develop` branch.
I attach here the database generated this weekend with the inverse reachability (rightArm + trunk):
[robot.4a0cf68d3eba8fe6aa3e92f28fa3c9ed.zip](https://github.com/roboticslab-uc3m/teo-openrave-models/files/1119017/robot.4a0cf68d3eba8fe6aa3e92f28fa3c9ed.zip)

rsantos88, 2017-07-03:
I append this updated info: `openravepy.databases.kinematicreachability: show, max reachability: 0.1388888888888889`

jgvictores, 2017-07-03:
I understand it just involves a few extra lines in `teo.robot.xml`. It would be nice to see these changes in `develop` (not the full .zip though).

rsantos88, 2017-07-04:
lol.. it's not the changes of `teo.robot.xml`. It's just the inverse-reachability database generated through the new manipulator `rightArm_trunk`. I've upload this database with the porpouse of see the results if you want without doing this process (you can read more [here](http://openrave.org/docs/0.8.0/openravepy/examples.tutorial_inversereachability/#optional-download-inverse-reachability-database)). You can also generate database but it takes to much time ;)

jgvictores, 2017-07-04:
Yes, what I meant is it would be good to merge [these](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/9f180ffd530aca90a3e601986d5cb14835bdebc8/openrave/teo/teo.robot.xml#L69-L75) and [these](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/9f180ffd530aca90a3e601986d5cb14835bdebc8/openrave/teo/teo.robot.xml#L91-L100) lines of `teo.robot.xml` from the `reachabilityWithTrunk` branch to `develop`.

rsantos88, 2017-07-04:
Ok, done it (#7)

jgvictores, 2017-07-04:
Great, thanks!
I haven't deleted the branch, in case there was anything you were still working on. Feel free to delete as required!

rsantos88, 2017-07-05:
Ok, it's already deleted :)

rsantos88, 2017-07-17:
I want to find a utility to this database, so I am trying first to follow the [inverse reachability tutorial](http://openrave.org/docs/0.8.0/openravepy/examples.tutorial_inversereachability/) but I've too many problems. I've written my issue [here](http://openrave-users-list.185357.n3.nabble.com/tutorial-inversereachability-error-td4027941.html), so if you know how to solve it I'll be very grateful :)

rsantos88, 2017-11-21:
I'll close this issue. , Currently, i am learning and using this database generator to generate a optimized and stable analytic inverse kinematics solvers for `rightArmTrunk` and `leftArmTrunk` and load them when we use a TEO model for OpenRave

jgvictores, 2017-11-21:
Could you point to the corresponding repositories, code, models?

jgvictores, 2018-04-04:
Reopening.

Will close after adding:
- Scripts in https://github.com/roboticslab-uc3m/teo-openrave-models/tree/develop/scripts/python
- Documentation in https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-openrave.md#generate-databases

jgvictores, 2019-11-20:
May be of relevance for this issue some day: Adopt <gripperjoints> tag regarding robot grippers (https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/86)

jgvictores, 2020-01-23:
> May be of relevance for this issue some day: Adopt tag regarding robot grippers ([roboticslab-uc3m/openrave-yarp-plugins#86](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/86))

I confirm the `<joints>` tag (aka `<gripperjoints>` afaik) is required for the OpenRAVE grasping module (`databases.grasping.GraspingModel(robot, target)`). Recovered for TEO at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/083998bfde1d05dcd96b7626da8a8c3bc8edf719

jgvictores, 2020-01-23:
BTW, added `horizontal` label to this issue. Should be split into issues per database generator, at least if we ever want to close it. ^^

jgvictores, 2020-01-23:
> Ok, I've done the script [here](https://github.com/roboticslab-uc3m/teo-openrave-models/commit/0acbba997185e1eaea9ee5b325358592c49f3b59)

Rebased and merged at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/1115cc4d78b283bb0b469f6865b7f007838bb8ef [ @rsantos88 thanks! ]

jgvictores, 2020-01-25:
Generated a cool `kinematicreachability` linked at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/673cdcada7fc3b43f95af61ec6eaa6c433145b0b, attaching screenshot below.

Some links from https://github.com/roboticslab-uc3m/teo-openrave-models/issues/3#issuecomment-315730184 and also some new ones:
- http://openrave-users-list.185357.n3.nabble.com/tutorial-inversereachability-error-td4027941.html
- https://www.google.com/search?q=KinematicReachability
- http://openrave.org/docs/0.8.0/openravepy/examples.tutorial_inversereachability/
- http://openrave-users-list.185357.n3.nabble.com/Kinematic-reachability-stats-and-reachability3D-td4026690.html

And, yes, as stated above: for `(...) kinematicreachability (...) --show`, `mayavi` is required. On this PC it was as easy as a `sudo -H pip2 install mayavi` (yep, using sudo and not on pip3 yet ftw).

[image]

jgvictores, 2020-01-25:
A `--show` on `inversereachability` seems to place an object at a random position, then show how to reach it.
[image]
[image]

jgvictores, 2020-01-27:
To sum up what we can see regarding http://openrave.org/docs/latest_stable/databases/#openrave-database-generators as of https://github.com/roboticslab-uc3m/teo-openrave-models/tree/e6401ac329d2c1716c95a6b4795560f7e60d7a40/scripts/bash#scriptsbash:
- `convexdecomposition` (:heavy_check_mark:): good (we expect to use this for collision again some day)
- `grasping`  (:white_check_mark:): good, but not introduced here as we already have https://github.com/roboticslab-uc3m/teo-grasp with cool examples. Note that as part of this study, at https://github.com/roboticslab-uc3m/teo-openrave-models/issues/44 we imported the grasping models into this repository.
- `inversekinematics` (:heavy_check_mark:): good (this is actually IKFast)
- `inversereachability` (:heavy_check_mark:): good (nice viz)
- `kinematicreachability` (:heavy_check_mark:): good (makes us see how limited we are, possibly due to joint limits)
- `linkstatistics` (:question:): only a small `.pp` file is generated, and we cannot viz.
- `visibilitymodel` (:x:): the generating script breaks

closed 2020-01-27 by jgvictores