roboticslab-uc3m/teo-openrave-models #4: Tool Center Points are not in the correct position

rsantos88, 2017-06-28:
As I said to @munozyanez yesterday, I've been checking the actuals TCP on openrave (using WorldRpcResponder) and `world whereis tcp (manipulator)` command that uses [GetEndEffector()](https://github.com/rsantos88/openrave-yarp-plugins/blob/worldRpcResponder/openraveplugins/OpenraveWorldRpcResponder/OpenraveWorldRpcResponder.cpp#L501) function.

The visual result is this:
[image]
We've been considering that is important to move correctly the end of the TCP and fix it in the code  to calculate correctly the inverse reachability.

munozyanez, 2017-06-28:
I had this issue making one ikfast (https://github.com/munozyanez/i-kinematics) interface for the legs.
I will fork develop branch and commit the solution I gave, so you can check it and see if it can do for the arms.
If solution is ok, we can merge on develop.

munozyanez, 2017-06-28:
Done, @rsantos88, you can try the TCP check, with the updated model in the repo : https://github.com/munozyanez/teo-openrave-models .
As side comment, you can also see that <Translation>0 0 0</Translation> is different in this file:
https://github.com/munozyanez/teo-openrave-models/blob/develop/openrave/teo/teo.robot.xml

munozyanez, 2017-06-28:
Above  "0 0 0"  means [<]Translation[>]0 0  0[<]/Translation[>] at xml openrave format.
Markdown seems to process [<] tags, so they don't show up above.

David-Estevez, 2017-06-28:
Just FYI:
You can make them appear by escaping them: \<Translation\> or putting them in a code environment `<Translation>`. Otherwise, markdown interprets them as  html / xml tags (so they are not rendered).

Good luck solving the issue! :four_leaf_clover: :four_leaf_clover:  :crossed_fingers:

munozyanez, 2017-06-28:
Yei, thank you for the info :-).

rsantos88, 2017-06-29:
Thanks @munozyanez. I've corrected the end of the left and right arm TCPs and I've checked the TCP of the legs. The changes of the code has been upload here (d48a6bb8d9f8e30d36c27e6dd1c11a35c7b653e4) and here (b9b67c059866f6288bbeab0c773ac173364ecd68), so we can merge it when you want. I also took the opportunity to change `joint circular` values to false for respect the joint limits.

The visual result is this:
[image]

munozyanez, 2017-06-29:
Looks great!
Thanks @rsantos88 for the work. Let's broadcast before merging and wait for comments.

rsantos88, 2017-06-30:
What do you think @jgvictores ? If you see that is OK we can merge this changes with the old model and then I can continue with this request (#3):

> @rsantos88 Do you think you could create 2 more manipulators? Something like fullRightArm and fullLeftArm that include the trunk (initially, including all joint limits).

jgvictores, 2017-06-30:
Yes, definitely! Looking forward to seeing that merged!! 😃

rsantos88, 2017-06-30:
Ok, here it is! (#6)

munozyanez, 2017-06-30:
Closing issue.

closed 2017-06-30 by munozyanez