roboticslab-uc3m/teo-openrave-models #8: Add the plate model to teo-openrave-models

RaulFdzbis, 2017-07-12:
The plate 3D model should be added to teo-openrave-models, in order to  be able to perform a correct simulation with the "waiter" demo. The idea is to generate a new teoSim model with the plate, or add the plate as a param to the current model. In order to do this some steps must be done:

- [ ] Add the plate model and make sure the new model works with teoSim (simulator).

- [ ] Check if the collision avoidance plugin works with the plate model.

jgvictores, 2017-07-12:
Bump @jmgarciah for awareness of this ongoing process. :-)

jgvictores, 2017-12-18:
Was in fact blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/36, which has now been closed. No longer blocked!

jgvictores, 2017-12-18:
Will only work with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/db1501fa14cc80b8a04750c7ffe1a2c50526a167/examples/openrave-YarpPluginLoader-controlboard.py
Blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/27 and thus by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47.

jgvictores, 2017-12-18:
PS: Will only work with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/cc6dac0ffc8b30d05132e54aebbb7abe1ce89fa2/examples/openraveYarpControlboard.py
Blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/27 and thus by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47.

RaulFdzbis, 2017-12-18:
I am not sure about the current state of this issue, since in the last meeting we have, this assignment was changed. I was working to export the TEO collision model to other platforms like [ARMARX](https://github.com/roboticslab-uc3m/tools/tree/develop/programs/openraveppToSTL) and so on.

@jmgarciah do you have any information about the state of this issue?.

jgvictores, 2017-12-18:
We have 3fa8e07 in this repo, on branch `teo-plate`. The thing is, it was branched out from current `develop-bad`, so suffers from same effects. I'd like to keep this issue open until it the related orphan branch is merged (currently blocked from merge, as explained above).

jgvictores, 2018-01-15:
Not blocked anymore.

jgvictores, 2018-01-15:
Related: #9

jgvictores, 2018-02-10:
Current models at http://robots.uc3m.es/svn/TEOrepo/mechanics/tray/

jgvictores, 2020-01-18:
A hack could be to use https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/0209324aaff63db8dd893d66d51729124c42ca9a/programs/openraveYarpWorldClient to load the model via https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/0209324aaff63db8dd893d66d51729124c42ca9a/libraries/OpenravePlugins/OpenraveYarpWorld, then use the second (maybe modifiy the first to do so) to do an `OpenRAVE`ish `Grab()`.