roboticslab-uc3m/teo-openrave-models #9: [teoSim] Implement hand joint

jgvictores, 2017-08-16:
_From @PeterBowman on August 16, 2017 10:19_

_From @jgvictores on May 25, 2015 10:6_

Left and right hand joints currently not implemented in simulator.

_Copied from original issue: roboticslab-uc3m/kinematics-dynamics#10_

_Copied from original issue: roboticslab-uc3m/openrave-yarp-plugins#32_

jgvictores, 2017-08-16:
_From @PeterBowman on August 16, 2017 10:20_

Keep in mind that teoSim is deprecated. Please update the title accordingly with whatever comes out of #27.

jgvictores, 2017-08-16:
Plugins use whatever are defined as the manipulator. Moving to https://github.com/roboticslab-uc3m/teo-openrave-models

PeterBowman, 2017-08-16:
I totally misinterpreted this, sorry. Let's get a more accurate title - I guess that the intention is to enable the open-close animation of both grabbers?

jgvictores, 2017-08-16:
Precisely. On the physical robot, it is implemented as the last joint of the arm kinematic chain.

jgvictores, 2017-10-04:
Update on this:
@rsantos88 started commenting on this on what is now https://github.com/roboticslab-uc3m/QA/issues/31
1. I'd leave https://github.com/roboticslab-uc3m/QA/issues/31 to provide a general answer to the question of "How to do a grasping hand for Teo?".
1. I'd work on this issue for "Enable LacqueyFetch open/close as joint of arm kinematic chain" via openrave model. 

Regarding this second point, I agree with @rsantos88 that "I think we need to separate the fingers in three different 3D files" is a good starting point, and this is the repo to work on!

rsantos88, 2017-10-05:
BTW ( @jgvictores writing ), the 3d model source (solidworks?) of the fetch hand was available from lackey's web site. We have a copy in one of our repos.

jgvictores, 2017-10-05:
Found it!
It's actually a STEP file, which is pretty managable from FreeCAD, @rsantos88 if you want to give it a try. 🙂 
http://robots.uc3m.es/svn/RH2repo/Modelling/Mechanical/Mano_Lacquey/

rsantos88, 2017-10-18:
Ok, I've uploaded the new TEO model that @smcdiaz given to me here with the fingers separated:
http://robots.uc3m.es/svn/TEOrepo/mechanics/Model/

rsantos88, 2017-10-19:
There are new changes in the SVN repo: [here](http://robots.uc3m.es/svn/TEOrepo/mechanics/Model/Conversions/Version%202017/VRML/Brazo_Der/) you can find a more simple model of torque sensor and a flange model between the wrist and the palm of the hand.

rsantos88, 2017-11-17:
It's ready!! This [pull-request](https://github.com/roboticslab-uc3m/teo-openrave-models/pull/12) adds the following improvements:
* Added new hands with separated and movable fingers. The original hand has been obtained from [teo.step](http://robots.uc3m.es/svn/TEOrepo/mechanics/Model/Conversions/Version%202017/STEP/TEO.STEP) model.
* recalculated the coordinates origin of all the hand parts on the axis of rotation of each joint
* created new kinbodys: `eftHand` and `rightHand` adapted for grasping usage with openrave
* added two more manipulators: (left hand + trunk) and (right hand +trunk)
* calculated inverse kinematic of: `leftArm`, `leftArmTrunk`, `rightArm`, `rightArmTrunk` and appended on `ik` directory and loaded them when we open the model for getting IK solutions.
* Other minor changes...

rsantos88, 2017-11-17:
@AlvaroMartinezR is documenting how he has calculated the origin of coordinates of each piece, in order to leave everything explained ;)

jgvictores, 2017-11-17:
> @AlvaroMartinezR is documenting how he has calculated the origin of coordinates of each piece, in order to leave everything explained ;)

Great!

AlvaroMartinezR, 2017-11-20:
I used CAE programs to obtain the key points of the hand, first I used Abaqus CAE but then changed to SolidWorks. To calculate these coordinates I went axis by axis to obtain all (x, y, z) dimensions for all points. 

I made a sketch providing coordinates. It was created with the idea to obtain the position of the next joint (in this case from the palm to the fingers), not to provide a complete descriptive sketch of the part.

If you think some distance is missing let me know and I will add it. with this measurements I was able to create a hand model able to move all fingers. Seen here:
[image]

I attach the sketches, let me know where they should be achieved.
[Falange_Proximal.PDF](https://github.com/roboticslab-uc3m/teo-openrave-models/files/1487404/Falange_Proximal.PDF)
[Mano_Laquey_Movil.PDF](https://github.com/roboticslab-uc3m/teo-openrave-models/files/1487405/Mano_Laquey_Movil.PDF)

jgvictores, 2017-11-21:
Please upload the `PDF` files to the https://github.com/roboticslab-uc3m/teo-hardware-manual repository, directly to the `assets` folder. Thanks!

AlvaroMartinezR, 2017-11-22:
Done, I created a separated folder in order to have a folder with several unrelated files.

jgvictores, 2017-11-22:
> Done, I created a separated folder in order to have a folder with several unrelated files.

@AlvaroMartinezR Thanks! Did some renaming and permission setting at https://github.com/roboticslab-uc3m/teo-hardware-manual/commit/11976112268db3f9501b3962eb941324974209a9

rsantos88, 2017-11-27:
I've created a new branch named [teo-grasp](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/teo-grasp) (cloned branch of old teo-fingers). That branch is easier to identify for grasping task related to https://github.com/roboticslab-uc3m/teo-grasp.
If all it's ok and correctly documented, we can close this issue.

AlvaroMartinezR, 2017-11-27:
Ok for me. :dancer:

jgvictores, 2017-11-27:
Ok, set as `blocked` upstream, as the model will be affected by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/36.
I've increased the priority of the issue, which as others is blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/29 that is soon incoming.

jgvictores, 2017-12-09:
Removed `upstream` because the blocking issue is part of this organization.

jgvictores, 2017-12-18:
No longer blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/36 !

jgvictores, 2017-12-18:
Will only work with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/db1501fa14cc80b8a04750c7ffe1a2c50526a167/examples/openrave-YarpPluginLoader-controlboard.py
Blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/27 and thus by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47.

jgvictores, 2017-12-18:
PS: Will only work with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/cc6dac0ffc8b30d05132e54aebbb7abe1ce89fa2/examples/openraveYarpControlboard.py
Blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/27 and thus by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47.

rsantos88, 2018-01-15:
Maybe we can merge `teo-grasp` model with the actual (located on `develop` branch).
And maybe we can advance in this issue.

jgvictores, 2018-01-15:
No blocking issue. Fact is branches are pretty messed up right now. Feel free to mix, reorganize and merge into one nice develop if you want.

jgvictores, 2018-01-15:
Related: #8

jgvictores, 2018-02-10:
Lots done at #16.

jgvictores, 2018-04-04:
Started (thus blocked by) https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/86 for the specific implementation of this.

jgvictores, 2018-04-04:
In fact, considering this done and setting https://github.com/roboticslab-uc3m/teo-main/issues/19 blocked by new issue.

closed 2018-04-04 by jgvictores