roboticslab-uc3m/teo-openrave-models #10: The teoSim model doesn't move

rsantos88, 2017-09-14:
With the changes of thas last [issue](https://github.com/roboticslab-uc3m/teo-openrave-models/commit/049d38f88d372acc39b9bce956be96b12bbe8f39), you can find problems using **teoSim**. 
The model doesn't move... so, you have to use master branch since until it is resolved...

jgvictores, 2017-09-14:
Ok, that actually makes since... The most probable scenario is that two controllers (maybe not in terms of OpenRAVE controllerBase, but anyway, two controllers...) are taking over the same arm joints.

I guess this only affects the arms, right (please feedback on head, trunk and legs!)?

jgvictores, 2017-09-19:
Okay, I'm going to move `master` up to 767ba9ab3e579c502b887f7a0c57621952ba1b48.
`develop` will not work with current https://github.com/roboticslab-uc3m/openrave-yarp-plugins but I think it's something we should check for on that side, hence I have opened https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/36 to track this.

jgvictores, 2017-09-19:
PS: Current `master` now working with `teoSim`.

jgvictores, 2017-10-05:
Blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/36

jgvictores, 2017-12-18:
No longer blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/36 !

jgvictores, 2017-12-18:
Will only work with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/db1501fa14cc80b8a04750c7ffe1a2c50526a167/examples/openrave-YarpPluginLoader-controlboard.py
Blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/27 and thus by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47.

jgvictores, 2017-12-18:
PS: Will only work with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/cc6dac0ffc8b30d05132e54aebbb7abe1ce89fa2/examples/openraveYarpControlboard.py
Blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/27 and thus by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47.