roboticslab-uc3m/teo-openrave-models #11: Dynamic Information about TEO

AlvaroMartinezR, 2017-11-08:
I have been working with SolidWorks and other CAE programs in order to develop an accurate dynamic model for TEO, including masses, inertias, CoM. It is not yet finished, but I wonder in which repository this kind of information (probably a .doc or a .pdf file) could be placed into.

PeterBowman, 2017-11-08:
These properties are parameters to the cartesian solver and usually go to [teo-configuration-files](https://github.com/roboticslab-uc3m/teo-configuration-files). Take a look at [share/kinematics/](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/e52590665a54cfbf33863f6e6264bdcc1df154a1/share/kinematics).

Probably we should move this to roboticslab-uc3m/teo-main#17.

jgvictores, 2017-11-08:
This issue was moved to roboticslab-uc3m/teo-main#21

closed 2022-06-04 by PeterBowman