roboticslab-uc3m/teo-openrave-models #13: Error opening wrl files in Meshlab

David-Estevez, 2018-01-11:
When I try to open any wrl file in Meshlab I get the following error:

```
Error encountered while loading file:
"/home/def/Repositories/teo-openrave-models/openrave/teo/models/0.0^cabeza_links.wrl"

File: /home/def/Repositories/teo-openrave-models/openrave/teo/models/0.0^cabeza_links.wrl
Error details: -- line 3 col 17: invalid NodeBodyElement
```

I am using Meshlab v1.3.2 64bit.

jgvictores, 2018-01-11:
Replicated error opening these VRML V1.0 files using Meshlab v1.3.2 64bit on a different machine. It is however capable of opening the kinect VRML V2.0 file.

Was capable of opening the files with FreeCAD, would that be useful for you?

PS: Not sure if should be considered a bug in the scope if this repository, as openrave is capable of opening the files.

PeterBowman, 2020-12-27:
For the record, the only COLLADA file in this repo was [teo.dae](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/6a70b45af9ad622e1b6fe9952e63620e53ff880c/openrave/teo/teo.dae), removed because of https://github.com/roboticslab-uc3m/teo-openrave-models/issues/32.

PeterBowman, 2020-12-29:
Done at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/6dc56d8364ca8f2c615c0238a3d7426d50f68e2a. Successfully tested via OpenRAVE on Ubuntu 18.04 (OR 0.9.0 w/ Coin viewer) and Ubuntu 20.04 (OR 0.54.0 w/ OSG viewer).

Now I can open these .wrl files with Meshlab and convert them to .dae via `meshlabserver -i part.wrl -o part.dae`.

closed 2020-12-29 by PeterBowman

jgvictores, 2020-12-30:
@PeterBowman Thank you so much for this!

For further reference, how were these VRML 2 files generated?

PeterBowman, 2020-12-30:
Sorry, I meant to add that and forgot it. FreeCAD was my choice: it allows you to open VRML1 files and export them to VRML2 afterwards.

jgvictores, 2021-01-05:
> My original goal was to open them with Blender, but it failed.

Turns out Blender cannot open the generated VRML2 (runs into something like https://developer.blender.org/T51334 `IndexError: list index out of range while importing wrl file`), however, it can open each individual `.dae` file generated currently at https://github.com/roboticslab-uc3m/teo-gazebo-models/tree/bbd7e33a2b0e6accdbaba5aef144fdbfa5b96136/models. : )

jgvictores, 2021-03-02:
> My original goal was to open them with Blender, but it failed.

For what it's worth, brand new, shiny: https://github.com/roboticslab-uc3m/teo-blender-models