roboticslab-uc3m/teo-openrave-models #19: Proposal: Delete IK .so from this repo, document how to generate

jgvictores, 2018-03-15:
Proposal: Delete IK .so from this repo, document how to generate (for instance, in https://github.com/roboticslab-uc3m/teo-openrave-models/blob/develop/doc/teo-openrave-models-install.md)

@rsantos88 Doable?

Blocks #18

rsantos88, 2018-03-19:
With [this](https://github.com/roboticslab-uc3m/teo-grasp/commit/940886eb6f394381c28462d3e1c6af6c4803365b) little improvement, it can autogenerate `ik` files without the need of save `.so` files in [teo-openrave-models](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/develop/openrave/teo/ik) repository. Only one disadvantage: it takes around 10 minutes to calculate it.
So, you can feel free to remove `ik` directory with the files, and edit `xml` with `<iksolver>` labels

rsantos88, 2018-03-20:
PR of this (#22)

jgvictores, 2018-04-04:
Working generator script at https://github.com/roboticslab-uc3m/teo-openrave-models/blob/6cd40916d8637043d1c2ff587f6322a8cab3ffdf/scripts/python/generateIkfast.py

Only issue was #23 for which the existing workaround is to downgrade sympy.

Will proceed to merge #22.

jgvictores, 2018-04-04:
Added https://github.com/roboticslab-uc3m/teo-openrave-models/blob/develop/doc/teo-openrave-models-install.md#generate-openrave-databases section

closed 2018-04-04 by jgvictores

jgvictores, 2018-04-04:
Alse here: https://github.com/roboticslab-uc3m/installation-guides/commit/0fc5817181bf3fa6eef4d07b18f5a70091297da4