roboticslab-uc3m/teo-openrave-models #23: generateIkfast.py breaks

jgvictores, 2018-03-23:
Tried twice, log:
```
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 5
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 7
Generating IK equation of: rightArm_trunk (one time computation)...
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 5
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 7
openravepy.databases.inversekinematics: generate, Generating inverse kinematics for manip rightArm_trunk: Transform6D [0, 1, 5, 6, 7, 8], precision=None, maxcasedepth=3 (this might take up to 10 min)
openravepy.databases.inversekinematics: generate, creating ik file /home/yo/.openrave/kinematics.96c0b1d7cc7831ea0b3b751b1a753790/ikfast0x1000004a.Transform6D.0_1_5_6_7_8_f4_9.cpp
openravepy.ikfast: forwardKinematicsChain, moved translation Matrix([[0, 0, 0]]) to right end
openravepy.ikfast: forwardKinematicsChain, moved translation Matrix([[0, 0, 0]]) to left end
openravepy.ikfast: forwardKinematicsChain, moved translation on intersecting axis Matrix([[0, 0, 0]]) to left
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, 483/2500]]
openravepy.ikfast: generateIkSolver, [[cos(j0), -sin(j0), 0, 0],[sin(j0), cos(j0), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, 1, 0],[0, -1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j1), -sin(j1), 0, 0],[sin(j1), cos(j1), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, -1, 0, -61/200],[0, 0, -1, -6573/25000]]
openravepy.ikfast: generateIkSolver, [[cos(j4), -sin(j4), 0, 0],[sin(j4), cos(j4), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[0, 0, 1, 0],[-1, 0, 0, 0],[0, -1, 0, 21/250]]
openravepy.ikfast: generateIkSolver, [[cos(j5), -sin(j5), 0, 0],[sin(j5), cos(j5), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[0, 0, 1, 329/1000],[0, -1, 0, 0],[1, 0, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j6), -sin(j6), 0, 0],[sin(j6), cos(j6), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, 1, 0],[0, -1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j7), -sin(j7), 0, 0],[sin(j7), cos(j7), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, -1, -101/500],[0, 1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j8), -sin(j8), 0, 0],[sin(j8), cos(j8), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, 1, 0],[0, -1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j9), -sin(j9), 0, 0],[sin(j9), cos(j9), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, 1, -1/5],[0, -1, 0, 0]]
openravepy.ikfast: solveFullIK_6D, ikfast 6d: [j0, j1, j5, j6, j7, j8]
openravepy.ikfast: iterateThreeIntersectingAxes, found 3 consecutive intersecting axes links[8:13], rotvars=[j6, j7, j8], translationvars=[j0, j1, j5]
openravepy.ikfast: _SimplifyRotationNorm, equation too complex to simplify for rot norm: -px**2 - 101*px*r00*sin(j9)/250 - 101*px*r01*cos(j9)/250 + 8673*px*sin(j0)/12500 + 61*px*sin(j1)*cos(j0)/100 - py**2 - 101*py*r10*sin(j9)/250 - 101*py*r11*cos(j9)/250 + 61*py*sin(j0)*sin(j1)/100 - 8673*py*cos(j0)/12500 - pz**2 - 101*pz*r20*sin(j9)/250 - 101*pz*r21*cos(j9)/250 + 61*pz*cos(j1)/100 + 10201*r00**2*cos(j9)**2/250000 - 10201*r00**2/250000 - 10201*r00*r01*sin(j9)*cos(j9)/125000 + 875973*r00*sin(j0)*sin(j9)/6250000 + 6161*r00*sin(j1)*sin(j9)*cos(j0)/50000 - 10201*r01**2*cos(j9)**2/250000 + 875973*r01*sin(j0)*cos(j9)/6250000 + 6161*r01*sin(j1)*cos(j0)*cos(j9)/50000 + 10201*r10**2*cos(j9)**2/250000 - 10201*r10**2/250000 - 10201*r10*r11*sin(j9)*cos(j9)/125000 + 6161*r10*sin(j0)*sin(j1)*sin(j9)/50000 - 875973*r10*sin(j9)*cos(j0)/6250000 - 10201*r11**2*cos(j9)**2/250000 + 6161*r11*sin(j0)*sin(j1)*cos(j9)/50000 - 875973*r11*cos(j0)*cos(j9)/6250000 + 10201*r20**2*cos(j9)**2/250000 - 10201*r20**2/250000 - 10201*r20*r21*sin(j9)*cos(j9)/125000 + 6161*r20*sin(j9)*cos(j1)/50000 - 10201*r21**2*cos(j9)**2/250000 + 6161*r21*cos(j1)*cos(j9)/50000 - 65710929/625000000
openravepy.ikfast: _SimplifyRotationNorm, equation too complex to simplify for rot norm: -px**2 - 101*px*r00*sin(j9)/250 - 101*px*r01*cos(j9)/250 + 8673*px*sin(j0)/12500 + 61*px*sin(j1)*cos(j0)/100 - py**2 - 101*py*r10*sin(j9)/250 - 101*py*r11*cos(j9)/250 + 61*py*sin(j0)*sin(j1)/100 - 8673*py*cos(j0)/12500 - pz**2 - 101*pz*r20*sin(j9)/250 - 101*pz*r21*cos(j9)/250 + 61*pz*cos(j1)/100 + 10201*r00**2*cos(j9)**2/250000 - 10201*r00**2/250000 - 10201*r00*r01*sin(j9)*cos(j9)/125000 + 875973*r00*sin(j0)*sin(j9)/6250000 + 6161*r00*sin(j1)*sin(j9)*cos(j0)/50000 - 10201*r01**2*cos(j9)**2/250000 + 875973*r01*sin(j0)*cos(j9)/6250000 + 6161*r01*sin(j1)*cos(j0)*cos(j9)/50000 + 10201*r10**2*cos(j9)**2/250000 - 10201*r10**2/250000 - 10201*r10*r11*sin(j9)*cos(j9)/125000 + 6161*r10*sin(j0)*sin(j1)*sin(j9)/50000 - 875973*r10*sin(j9)*cos(j0)/6250000 - 10201*r11**2*cos(j9)**2/250000 + 6161*r11*sin(j0)*sin(j1)*cos(j9)/50000 - 875973*r11*cos(j0)*cos(j9)/6250000 + 10201*r20**2*cos(j9)**2/250000 - 10201*r20**2/250000 - 10201*r20*r21*sin(j9)*cos(j9)/125000 + 6161*r20*sin(j9)*cos(j1)/50000 - 10201*r21**2*cos(j9)**2/250000 + 6161*r21*cos(j1)*cos(j9)/50000 - 65710929/625000000
openravepy.ikfast: _SimplifyRotationNorm, equation too complex to simplify for rot norm: -px**2 - 101*px*r00*sin(j9)/250 - 101*px*r01*cos(j9)/250 + 8673*px*sin(j0)/12500 + 61*px*sin(j1)*cos(j0)/100 - py**2 - 101*py*r10*sin(j9)/250 - 101*py*r11*cos(j9)/250 + 61*py*sin(j0)*sin(j1)/100 - 8673*py*cos(j0)/12500 - pz**2 - 101*pz*r20*sin(j9)/250 - 101*pz*r21*cos(j9)/250 + 61*pz*cos(j1)/100 + 10201*r00**2*cos(j9)**2/250000 - 10201*r00**2/250000 - 10201*r00*r01*sin(j9)*cos(j9)/125000 + 875973*r00*sin(j0)*sin(j9)/6250000 + 6161*r00*sin(j1)*sin(j9)*cos(j0)/50000 - 10201*r01**2*cos(j9)**2/250000 + 875973*r01*sin(j0)*cos(j9)/6250000 + 6161*r01*sin(j1)*cos(j0)*cos(j9)/50000 + 10201*r10**2*cos(j9)**2/250000 - 10201*r10**2/250000 - 10201*r10*r11*sin(j9)*cos(j9)/125000 + 6161*r10*sin(j0)*sin(j1)*sin(j9)/50000 - 875973*r10*sin(j9)*cos(j0)/6250000 - 10201*r11**2*cos(j9)**2/250000 + 6161*r11*sin(j0)*sin(j1)*cos(j9)/50000 - 875973*r11*cos(j0)*cos(j9)/6250000 + 10201*r20**2*cos(j9)**2/250000 - 10201*r20**2/250000 - 10201*r20*r21*sin(j9)*cos(j9)/125000 + 6161*r20*sin(j9)*cos(j1)/50000 - 10201*r21**2*cos(j9)**2/250000 + 6161*r21*cos(j1)*cos(j9)/50000 - 65710929/625000000
openravepy.ikfast: _SimplifyRotationNorm, equation too complex to simplify for rot norm: -px**2 - 101*px*r00*sin(j9)/250 - 101*px*r01*cos(j9)/250 + 8673*px*sin(j0)/12500 + 61*px*sin(j1)*cos(j0)/100 - py**2 - 101*py*r10*sin(j9)/250 - 101*py*r11*cos(j9)/250 + 61*py*sin(j0)*sin(j1)/100 - 8673*py*cos(j0)/12500 - pz**2 - 101*pz*r20*sin(j9)/250 - 101*pz*r21*cos(j9)/250 + 61*pz*cos(j1)/100 + 10201*r00**2*cos(j9)**2/250000 - 10201*r00**2/250000 - 10201*r00*r01*sin(j9)*cos(j9)/125000 + 875973*r00*sin(j0)*sin(j9)/6250000 + 6161*r00*sin(j1)*sin(j9)*cos(j0)/50000 - 10201*r01**2*cos(j9)**2/250000 + 875973*r01*sin(j0)*cos(j9)/6250000 + 6161*r01*sin(j1)*cos(j0)*cos(j9)/50000 + 10201*r10**2*cos(j9)**2/250000 - 10201*r10**2/250000 - 10201*r10*r11*sin(j9)*cos(j9)/125000 + 6161*r10*sin(j0)*sin(j1)*sin(j9)/50000 - 875973*r10*sin(j9)*cos(j0)/6250000 - 10201*r11**2*cos(j9)**2/250000 + 6161*r11*sin(j0)*sin(j1)*cos(j9)/50000 - 875973*r11*cos(j0)*cos(j9)/6250000 + 10201*r20**2*cos(j9)**2/250000 - 10201*r20**2/250000 - 10201*r20*r21*sin(j9)*cos(j9)/125000 + 6161*r20*sin(j9)*cos(j1)/50000 - 10201*r21**2*cos(j9)**2/250000 + 6161*r21*cos(j1)*cos(j9)/50000 - 65710929/625000000
openravepy.ikfast: _SimplifyRotationNorm, equation too complex to simplify for rot norm: -px**2 - 101*px*r00*sin(j9)/250 - 101*px*r01*cos(j9)/250 + 8673*px*sin(j0)/12500 + 61*px*sin(j1)*cos(j0)/100 - py**2 - 101*py*r10*sin(j9)/250 - 101*py*r11*cos(j9)/250 + 61*py*sin(j0)*sin(j1)/100 - 8673*py*cos(j0)/12500 - pz**2 - 101*pz*r20*sin(j9)/250 - 101*pz*r21*cos(j9)/250 + 61*pz*cos(j1)/100 + 10201*r00**2*cos(j9)**2/250000 - 10201*r00**2/250000 - 10201*r00*r01*sin(j9)*cos(j9)/125000 + 875973*r00*sin(j0)*sin(j9)/6250000 + 6161*r00*sin(j1)*sin(j9)*cos(j0)/50000 - 10201*r01**2*cos(j9)**2/250000 + 875973*r01*sin(j0)*cos(j9)/6250000 + 6161*r01*sin(j1)*cos(j0)*cos(j9)/50000 + 10201*r10**2*cos(j9)**2/250000 - 10201*r10**2/250000 - 10201*r10*r11*sin(j9)*cos(j9)/125000 + 6161*r10*sin(j0)*sin(j1)*sin(j9)/50000 - 875973*r10*sin(j9)*cos(j0)/6250000 - 10201*r11**2*cos(j9)**2 [... cut ...]
    if e.has(I) or isinf(e) or isnan(e):
  File "/usr/lib/python2.7/dist-packages/sympy/core/expr.py", line 224, in __float__
    raise TypeError("can't convert complex to float")
TypeError: can't convert complex to float
```

I recall that convexdecomposition autogenerate didn't work,  we ended up calling generate and save.

@rsantos88 Any ideas for the current [generateIkfast.py](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/001be5f1d13d25df8845a9fd598108fa9114ce03/scripts/python/generateIkfast.py)?

jgvictores, 2018-04-03:
Blocks #19

rsantos88, 2018-04-04:
Apparently working at [current master](https://github.com/rdiankov/openrave/commit/7c5f5e27eec2b2ef10aa63fbc519a998c276f908).

FYI (petition by @jgvictores ):
```
adium-theme-ubuntu (0.3.4)
apptools (4.1.0)
apt-xapian-index (0.45)
argparse (1.2.1)
astroid (1.0.1)
chardet (2.0.1)
colorama (0.2.5)
command-not-found (0.3)
configobj (4.7.2)
debtagshw (0.1)
decorator (3.4.0)
defer (1.0.6)
dirspec (13.10)
docutils (0.11)
dulwich (0.9.4)
duplicity (0.6.23)
envisage (4.1.0)
f.lux-indicator-applet (1.1.11-pre)
fastimport (0.9.2)
h5py (2.7.1)
html5lib (0.999)
httplib2 (0.8)
ipython (1.2.1)
Jinja2 (2.7.2)
lockfile (0.8)
logilab-common (0.61.0)
lxml (3.3.3)
MarkupSafe (0.18)
matplotlib (1.3.1)
mayavi (4.1.0)
mercurial (2.8.2)
numpy (1.13.3)
oauthlib (0.6.1)
oneconf (0.3.7.14.04.1)
openravepy (0.9.0)
PAM (0.4.2)
pandas (0.21.0)
pep8 (1.4.6)
pexpect (3.1)
Pillow (2.3.0)
pip (1.5.4)
piston-mini-client (0.7.5)
Pivy (0.5.0)
psutil (1.2.1)
pycrypto (2.6.1)
pycups (1.9.66)
pycurl (7.19.3)
pyface (4.1.0)
pyflakes (0.8.1)
Pygments (1.6)
pygobject (3.12.0)
pygpgme (0.3)
pylint (1.1.0)
pyOpenSSL (0.13)
pyparsing (2.0.1)
pyserial (2.6)
pysmbc (1.0.14.1)
pysqlite (1.0.1)
python-apt (0.9.3.5ubuntu2)
python-dateutil (2.6.1)
python-debian (0.1.21-nmu2ubuntu2)
pytz (2017.3)
pyxdg (0.25)
pyzmq (14.0.1)
rabbitvcs (0.15.2)
reportlab (3.0)
requests (2.2.1)
roman (2.0.0)
rope (0.9.2)
scipy (1.0.0)
sessioninstaller (0.0.0)
setuptools (3.3)
simplegeneric (0.8.1)
simplejson (3.3.1)
six (1.5.2)
software-center-aptd-plugins (0.0.0)
Sphinx (1.2.2)
spyder (2.2.5)
sympy (0.7.1)
system-service (0.1.6)
traits (4.1.0)
traitsui (4.1.0)
Twisted-Core (13.2.0)
Twisted-Web (13.2.0)
unity-lens-photos (1.0)
urlgrabber (3.9.1)
urllib3 (1.7.1)
vboxapi (1.0)
VTK (5.8.0)
wheel (0.24.0)
wsgiref (0.1.2)
wxPython (2.8.12.1)
wxPython-common (2.8.12.1)
xdiagnose (3.6.3build2)
yum-metadata-parser (1.1.4)
zope.interface (4.0.5)
```

rsantos88, 2018-04-04:
I normally use this scripts to install openrave and its dependences: https://github.com/crigroup/openrave-installation (if it were useful)

jgvictores, 2018-04-04:
> I normally use this scripts to install openrave and its dependences: https://github.com/crigroup/openrave-installation (if it were useful)

Yes, that is useful information.

rsantos88, 2018-04-04:
*note for the flies:* in my computer I have not installed `OpenSceneGraph`

jgvictores, 2018-04-04:
My main suspect related to those guidelines and the error described above is currently [here](https://github.com/crigroup/openrave-installation/blob/83376312a3ba2676d98eb52ef7f0e6267213e85b/install-openrave.sh#L9-L13): `Downgrading sympy to version 0.7.1...`.
Perhaps somewhere between sympy 0.7.1 and xenial sympy 0.7.6 they introduced a `from math import *` as described [here](https://stackoverflow.com/questions/38730671/error-with-pow-cant-convert-complex-to-float).

jgvictores, 2018-04-04:
Downgrading sympy to version 0.7.1 allows ikfast generation!

Going to upgrade to 1.1.1, perhaps it's due to a bug in 0.7.6.

jgvictores, 2018-04-04:
Unfortunately, same old error with sympy 1.1.1.

Downgrading to 0.7.1 again, closing issue, and commenting at https://github.com/roboticslab-uc3m/installation-guides/issues/41 or https://github.com/roboticslab-uc3m/installation-guides in general

@rsantos88 Thanks for the help to determine this bug/error!

jgvictores, 2018-04-04:
...and a corresponding open PR, https://github.com/rdiankov/openrave/pull/407 that aspires for sympy 1.0 but for now recommends 0.7.1

closed 2022-06-04 by PeterBowman