roboticslab-uc3m/teo-openrave-models #26: Update joint directions and limits

jgvictores, 2019-03-27:
Mostly updated at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/118f4725c86472dc1c8c3782cdde3d49856c6f74
- Directions: https://github.com/roboticslab-uc3m/teo-main/issues/39
- Limits: https://github.com/roboticslab-uc3m/teo-main/issues/15

jgvictores, 2019-03-27:
Done at https://github.com/roboticslab-uc3m/teo-openrave-models/pull/25, merged at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/92c99a5af125b55b4636e51c30b6524c6bfa5977 [ thanks @rsantos88 !! ]

PeterBowman, 2019-06-26:
Reopening per https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-505823202.

PeterBowman, 2019-07-03:
Done by @rsantos88 at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/c4205a894506244a6d449c4251e374dc12e3380f, tested at https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-508151812 and hotfixed into develop.

closed 2019-07-03 by PeterBowman