roboticslab-uc3m/teo-openrave-models #28: Update forearm link

PeterBowman, 2019-06-19:
The forearm is now 0.215 meters long, XMLs still reflect 0.202: [left arm](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/435b8cce063d90fbe0b1943aca918c05e3f75aa6/openrave/teo/teo_leftArm.kinbody.xml#L105), [right arm](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/435b8cce063d90fbe0b1943aca918c05e3f75aa6/openrave/teo/teo_rightArm.kinbody.xml#L105).

Originates from https://github.com/roboticslab-uc3m/teo-main/issues/23.

jgvictores, 2019-11-30:
At https://github.com/roboticslab-uc3m/teo-openrave-models/issues/42 "Trace and document origin of `.wrl` models" I saw there are two folders within `/TEOrepo/mechanics/Model/Conversions`:
- `Version 2016`
- `Version 2017`

Seeing that https://github.com/roboticslab-uc3m/teo-main/issues/23 "Changes of forearm and its consequences" is from 2017... @smcdiaz Do you know if these facts are related? Does the transition between `Version 2016` and `Version 2017` relate to the changes of the forearm?

jgvictores, 2019-12-02:
> Seeing that [roboticslab-uc3m/teo-main#23](https://github.com/roboticslab-uc3m/teo-main/issues/23) "Changes of forearm and its consequences" is from 2017... @smcdiaz Do you know if these facts are related? Does the transition between `Version 2016` and `Version 2017` relate to the changes of the forearm?

Nope. Spoke with @smcdiaz. Differences are explained in the corresponding README files, duh.

smcdiaz, 2019-12-03:
Done but where to upload it?

jgvictores, 2019-12-03:
How about overwriting http://robots.uc3m.es/svn/TEOrepo/mechanics/Model/ and creating a `http://robots.uc3m.es/svn/TEOrepo/mechanics/Model/Conversions/Version 2019` ?

jgvictores, 2020-01-11:
Ok http://robots.uc3m.es/svn/TEOrepo/mechanics/Model/Conversions/Version%202019/ [thanks @smcdiaz]!!

PeterBowman, 2020-12-29:
> Ok http://robots.uc3m.es/svn/TEOrepo/mechanics/Model/Conversions/Version%202019/ [thanks @smcdiaz]!!

Sorry, but I don't see any change, link length is still 202 mm in both STEP and VRML files (checked on right arm).

PeterBowman, 2020-12-29:
Sorry again, but none of these (2016, 2017, 2019) correspond to the current VRML model. For instance, the legs look quite different... Has anyone worked on the .wrl files uploaded to this repo (or wherever else, but not on the original SolidWorks project) since 2016?

smcdiaz, 2020-12-29:
@rsantos88 is working with the proper models and he updated it recently. Anyway, I will check the models

PeterBowman, 2021-01-07:
I have just taken a look at FrontalLeftElbow.wrl, there are lots of hidden vertices/faces. After removing and exporting them with Meshlab, the overall file size dropped almost by a half. That's just one .wrl.

rsantos88, 2021-01-21:
> @rsantos88 is working with the proper models and he updated it recently. Anyway, I will check the models

Sorry! I haven't seen this comment before!
In my case, to carry out my experiments with Blender and the mocap system, I started from the most detailed model that I found with the real dimensions of the robot. This model contains all the parts that make up Teo in a single file. It's located inside the SVN repository at Conversions/Version 2017/VRML/TEO/TEO.wrl in VRML V2.0 format. In this model, I have measured the forearm piece, obtaining 5.62412 cm

[image]

After searching the 2019 version, I found that the forearm measurement was 6 cm (@smcdiaz  confirmed that this was the only change compared to the 2017 version). So I extruded the piece to 6 cm.

[image]

> Sorry again, but none of these (2016, 2017, 2019) correspond to the current VRML model. For instance, the legs look quite different... Has anyone been working on the .wrl files uploaded to this repo (or wherever else, but not on the original SolidWorks project) since 2016?

In my case, I haven't touched or made any changes about the SVN repository

PeterBowman, 2021-01-21:
Thank you. So this is even weirder than I thought. The difference of forearm lengths varies between ~3.76 mm and 13 mm depending on the model.

> t's located inside the SVN repository at Conversions/Version 2017/VRML/TEO/TEO.wrl in VRML V2.0 format.

Don't trust the year, this model was generated in 2015 according to the initial lines:

```
# VRML file: F:\Repositorio TEO\Modelling\Mechanical\Model\Conversions\VRML\TEO\TEO\TEO.wrl
#
# Converted by the PolyTrans geometry converter from Okino Computer Graphics, Inc.
# Date/time of export: 01/22/2015 18:46:25
#
# Bounding box of geometry = (-0.178,-0.844897,-0.39392) to (0.174583,0.7857,0.39392).
```

It also lacks some level of detail in the ankle joints that makes it clearly distinct from current OpenRAVE/VRML models.

> In my case, I haven't touched or made any changes about the SVN repository

I was talking about the [teo-openrave-models](https://github.com/roboticslab-uc3m/teo-openrave-models) repository.

rsantos88, 2021-01-22:
> It also lacks some level of detail in the ankle joints that makes it clearly distinct from current OpenRAVE/VRML models.

yes, totally true ...
I've also found another difference although imperceptible at first.

[image]

In the model I'm working with, the piece that corresponds to the shoulders produces a slight asymmetry in the position of the arms. This is because the piece has been created with a certain displacement with respect to the axis of rotation. If I isolate both pieces, I can see that the second is the result of having duplicated the first and rotated in the Z axis 180 degrees. This produces a forward displacement of one arm relative to the other.

[image]

In the OpenRAVE / VRML models I've observed that this doesn't occur ...