roboticslab-uc3m/teo-openrave-models #29: Fix units in velocity limits

PeterBowman, 2019-06-28:
I noticed that velocity limits are a bit too high:

```
$ yarp rpc /teoSim/leftLeg/rpc:i
>> get vlim 0
Response: [is] vlim -1718.87338539247 1718.87338539247 [tsta] 0 1561730960.70992 [ok]
```

Those values are fetched from the `<maxvel>` tags, e.g. [here](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/435b8cce063d90fbe0b1943aca918c05e3f75aa6/openrave/teo/teo_leftLeg.kinbody.xml#L27). But OR parses them in radians/second, hence the madness. I'd use `<maxveldeg>` per [Format:XML](http://openrave.programmingvision.com/wiki/index.php/Format:XML) (actually works) and tweak the limits accordingly, e.g. to 30 deg/s or a bit higher.

@jgvictores most kinbody description files set 3 rad/s, leg XMLs use 30 rad/s. Is 30 deg/s OK for **all** robot joints (including Dextra fingers)? Current (recently reviewed) default is 10 deg/s ([example](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/f1c5cb41c9993fea3611342036f7e0dfd6ffc192/share/launchManipulation/conf/launchManipulation.ini#L11)), which is a bit too low for highly dynamic tasks, e.g. walking or bimanipulation.

jgvictores, 2019-06-28:
Cool, thanks a lot for detecting this!

PeterBowman, 2019-06-28:
Done, also fixed in AMOR and ASIBOT OpenRAVE models.

closed 2019-06-28 by PeterBowman