roboticslab-uc3m/teo-openrave-models #31: Reorganizing TEO openrave model

rsantos88, 2019-11-08:
Following <http://openrave.programmingvision.com/wiki/index.php/Format:XML#Joining_A_Robot_Arm_and_Hand>, we would like:
- [ ] A model of TEO without hands/grippers (OpenRAVE robot)
- [ ] Models of different TEO hands/grippers (also OpenRAVE robots)
    - [ ] Fetch
    - [ ] Dextra
- [ ] Models that integrate the above
    - [ ] TEO + Fetch (rotation possibility 1)
    - [ ] TEO + Fetch (rotation possibility 2)
    - [ ] TEO + Dextra

rsantos88, 2019-11-08:
Done at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/a27e6b9e8708e52b198ebea80c6d3579b419978d !!!

closed 2019-11-08 by rsantos88

jgvictores, 2019-11-11:
PS: `teoSim` was consequently updated at https://github.com/roboticslab-uc3m/teo-configuration-files/commit/548f608f0863936a0cbb3e3d2521c177915837e3