roboticslab-uc3m/teo-openrave-models #32: Replace teo.dae with script to generate it

rsantos88, 2019-11-08:
Replace teo.dae with script to generate it, because we do not even know where it is coming from.

jgvictores, 2019-11-14:
* Note: I was about to delete it because my OpenRAVE cannot even open it (currently at https://github.com/roboticslab-uc3m/teo-openrave-models/blob/9c8ab84d78b5dcf90e25a7f412011597288818eb/openrave/teo/teo.dae), but then I remembered I believe COLLADA can only be open with the OSG viewer plugin (not the Qt one I use).

jgvictores, 2019-11-19:
Deleted at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/176e7a4e325e953da4d52e1042cdae09c4c409bd

jgvictores, 2019-12-02:
Turns out this is as easy as:
- <http://openrave.org/docs/0.8.2/robots_overview/>:
```bash
openrave -save myrobot.zae myrobot.xml
```
- <http://openrave-users-list.185357.n3.nabble.com/Creating-COLLADA-file-td4025650.html>:
```
openrave -save myrobot.dae myrobot.xml
```

Note that you may not be able to open the file with OpenRAVE (I believe it must be compiled with OSG). However, I was able to load `openrave -save teo_dextra.robot.dae teo_dextra.robot.xml` on Gazebo via:
```xml
<?xml version="1.0"?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>
    <model name="my_mesh">
      <pose>0 0 1.0  0 0 0</pose>
      <static>true</static>
      <link name="body">
        <visual name="visual">
          <geometry>
            <mesh><uri>file://teo.robot.dae</uri></mesh>
          </geometry>
        </visual>
      </link>
    </model>
  </world>
</sdf>
```

jgvictores, 2019-12-02:
Scripts added at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/358ddcc067dec62d0034b5a3b5e27926168651bd

jgvictores, 2019-12-02:
> Note that you may not be able to open the file with OpenRAVE (I believe it must be compiled with OSG).

Nope: sudo apt install libcollada-dom2.4-dp-dev  # Open .zae files, only Ubuntu 16.04

closed 2019-12-02 by jgvictores

jgvictores, 2019-12-02:
Updated info at http://robots.uc3m.es/gitbook-installation-guides/install-openrave.html#generate-databases via https://github.com/roboticslab-uc3m/installation-guides/commit/d925179977b6986a20204e073b6d22b3f8bf2d8d