roboticslab-uc3m/teo-openrave-models #37: Rename model files according to Unified TEO link names

jgvictores, 2019-11-13:
Rename model files according to Unified TEO link names, as a followup of https://github.com/roboticslab-uc3m/teo-openrave-models/pull/36. Update XMLs to match.

rsantos88, 2019-11-14:
In order to rename the hands, I would like to be clear about the best name. I've thought:
* For lacqueyFetch: 
`hand_frontal_wrist.wrl` -> `FrontalWristHand.wrl`
`hand_phalange_distal.wrl` -> `DistalPhalangeHand.wrl`
`hand_phalange_proximal.wrl` -> `ProximalPhalangeHand.wrl`

* For Dextra, you can propose (and edit this issue) to avoid name mistakes:
`6.1^mano_abductor_links.wrl` -> ?
`6.1^mano_static_links.wrl` -> ?
`6.2^mano_abductor_links.wrl` -> ?
`6.2^mano_static_links.wrl` -> ?
`6.3^mano_proximal_links.wrl` -> ?
`6.3^mano_static_links.wrl` -> ?
`6.4^mano_middle_links.wrl` -> ?
`6.5^mano_distal_links.wrl` -> ?
`6.6^mano_proximalthumb_links.wrl` -> ?
`6.7^mano_distalthumb_links.wrl` -> ?
`7.1^manoizq_static_links.wrl` -> ?
`muneca.wrl` -> ?

rsantos88, 2019-11-14:
For IacqueyFetch it can be better:
`hand_frontal_wrist.wrl` -> `FrontalWristLacquey` or `FrontalWristLaqueyFetch` ?

rsantos88, 2019-11-14:
For Dextra it can be a similar as a `body name` like:
`6.3^mano_static_links.wrl`->`FrontalRightWristDextra.wrl` in relation to https://github.com/roboticslab-uc3m/teo-openrave-models/blob/develop/openrave/teo/rightDextra.kinbody.xml#L5-L14

jgvictores, 2019-11-14:
I propose re-structuring the folder structure in [teo-openrave-models/openrave](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/1916d1c19244bce8b7c930b3c3193e78593a02a8/openrave). Currently we only see `teo`, but this can be split into `teo`+`lacqueyFetch`+`dextra`.
This can be done in a separate issue if you prefer. For now, how about making `openrave/teo/models/lacqueyFetch` and `openrave/teo/models/dextra` and naming files simply `FrontalWrist.wrl` or `FrontalRightWrist.wrl` in their corresponding folder?

rsantos88, 2019-11-14:
I've re-structured the models directory with: 
* `openrave/teo/models/lacqueyFetch`
* `openrave/teo/models/dextra`
* `openrave/teo/models/teo`

https://github.com/roboticslab-uc3m/teo-openrave-models/tree/rename-model-files/openrave/teo/models

rsantos88, 2019-11-14:
some of the names of the **Dexta** hand are repeated, except for the enumeration that heads the name, for example `6.1^mano_static_links.wrl`,` 6.2^mano_static_links.wrl` and `6.3^mano_static_links.wrl`. What name do you want to put to differentiate it?

jgvictores, 2019-11-14:
I'd simply append (on right side) the `XXXXX_1.wrl` vs `XXXXX_2.wrl` vs `XXXXX_3.wrl`.

jgvictores, 2019-11-14:
> I'd simply append (on right side) the `XXXXX_1.wrl` vs `XXXXX_2.wrl` vs `XXXXX_3.wrl`.

Unless more semantically accurate names could be given.

rsantos88, 2019-11-14:
> I'd simply append (on right side) the `XXXXX_1.wrl` vs `XXXXX_2.wrl` vs `XXXXX_3.wrl`.

I have renamed the files in [this way](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/rename-model-files/openrave/teo/models/dextra) (this part is the one I most doubt, it would be logical to give it more specific names). Regarding the file [muneca.wrl](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/rename-model-files/openrave/teo/models/dextra/muneca.wrl), it seems that it is not called at any time. I left it just in case

rsantos88, 2019-11-14:
Ready to merge https://github.com/roboticslab-uc3m/teo-openrave-models/pull/40 if you see it right, @jgvictores

jgvictores, 2019-11-14:
Perfect! Merged https://github.com/roboticslab-uc3m/teo-openrave-models/pull/40.

Announcement: after the merge I did the restructuring commented at https://github.com/roboticslab-uc3m/teo-openrave-models/issues/37#issuecomment-553833878 (via https://github.com/roboticslab-uc3m/teo-openrave-models/commit/9c8ab84d78b5dcf90e25a7f412011597288818eb), so please `git pull` and you'll see the new folder structure. ^^

Closing issue! @rsantos88 Thanks!

closed 2019-11-14 by jgvictores