roboticslab-uc3m/teo-openrave-models #38: Review "TCP" links in TEO model

jgvictores, 2019-11-13:
At https://github.com/roboticslab-uc3m/teo-openrave-models/commit/320b0f4fbcb56cebd6a7eea2f31d29abfae76829 for https://github.com/roboticslab-uc3m/teo-openrave-models/pull/36 i discovered some redundant transformations that may have meant sense in the past but now may not.

Here's an example (and there are 2 or 4 in total), compare:
- https://github.com/roboticslab-uc3m/teo-openrave-models/blob/1916d1c19244bce8b7c930b3c3193e78593a02a8/openrave/teo/teo_leftArm.kinbody.xml#L149
- https://github.com/roboticslab-uc3m/teo-openrave-models/blob/1916d1c19244bce8b7c930b3c3193e78593a02a8/openrave/teo/teo_lacqueyFetch.robot.xml#L25

It can be seen that they are explicitly designed to compensate each other. Must rethink. Also related to https://github.com/roboticslab-uc3m/openrave-yarp-plugins

rsantos88, 2019-11-14:
> It can be seen that they are explicitly designed to compensate each other.

Yes, that seems. I will correct it when the issue https://github.com/roboticslab-uc3m/teo-openrave-models/issues/37 are finished

jgvictores, 2019-11-14:
Marking as `blocked`, as this still needs some thought and is related to https://github.com/roboticslab-uc3m/openrave-yarp-plugins [libraries/YarpPlugins/YarpOpenraveControlboard](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/9f4f66007f913025d9378ef9692a03999c5ddafc/libraries/YarpPlugins/YarpOpenraveControlboard).

jgvictores, 2019-11-30:
Some history:
- https://github.com/roboticslab-uc3m/teo-openrave-models/issues/4 "Tool Center Points are not in the correct position"
- https://github.com/roboticslab-uc3m/teo-openrave-models/pull/6 "Tool Center Points fixed to a correct position"